MRoboSub/mrobosub #22: Improve camera latency HSV/ML

alexanderbowler, 2024-08-25:
Will allow sub to move faster without having to wait and make sure camera is aligned with current time

RCoder01, 2025-01-26:
I looked into this a bit for ML, it seems like the latency is almost directly proportional to how many pixels/second we are requesting through the pipeline. Ie. reducing the frame rate or image size seems to reduce the latency. I would imagine the same is true of the HSV pipeline but I haven't tested that.

brandmcd, 2025-09-16:
Possibly look into exposure settings (lower exposure, raise gain) this has some info: https://github.com/opencv/opencv/issues/9738

The current code in botcam.py has 
`self.cap.set(cv2.CAP_PROP_AUTO_EXPOSURE, 1)
self.cap.set(cv2.CAP_PROP_EXPOSURE, 1000)`
which does not make sense with units?

alexanderbowler, 2025-09-17:
This looks promising if this is something you want to look into we can try some things out and then test on the sub