MRoboSub/mrobosub #43: Constant logging

RCoder01, 2024-11-14:
Add constant periodic logging for info like motor output, voltage(?), battery(?), temps, state machine state, running nodes, active topics, sensor data/predicted pose, etc.

Once we move to the SSD on the Jetson, we shouldn't have as many issues with storage so saving all this info should be feasible. If we find this takes too much storage, we may need to implement better compression schemes or delete/backup data after some time.

Primary motivation is yesterday's MHL 🔥

TongXinUmich, 2025-01-26:
added a logger to track one topic in logger branch. working on state machine state next.

imzaynb, 2025-09-10:
@TongXinUmich Do you have any update on this. Do you think you can complete this or do you want to transfer your progress to someone else. I'm totally down either way. Thanks!!