MRoboSub/mrobosub #67: Thruster control board restarts occassionally

RCoder01, 2025-03-26:
When driving the sub around at testing on 3/25/25, twice the thruster control board seemed to restart and cause a loss of control.

Ideally we eliminate this issue electrically, but worst case we should try to account for it in software.

RCoder01, 2025-04-19:
Didn't seem to happen at all during testing 4/18/25

RCoder01, 2025-09-10:
This is about the thruster control board (Polulu) which seemed to restart when pushing too much current to the thrusters. The temperature controller was the jetson restarting. At comp we just decreased the current limit but we had to drop it quite low k.

imzaynb, 2025-09-10:
Understood. Let's look into this. I'll move this to this year's project.